arduino-libraries/ArduinoModbus #5: Invalid CRC error

albydnc, 2018-10-20:
I'm using the latest version of the library. Via the rs485 library, I can get my counter to answer the request correctly. Implementing the sketch with this lib, it gives me Invalid CRC error and Response not from requested slave.
So there is a problem with the receiving buffer. Could it be timeout?  I think it is splitting the msg in two.

HW:
Arduino MKR WIFI1010 with MKR485 shield
Finder 7E.64.8.230.0210 Meter

sandeepmistry, 2018-10-22:
Hi @albydnc,

Could you please share the sketch you are using? Currently the library has a 1 second time out for RTU.

Would you please able to change the library here to see what data is being received?

https://github.com/arduino-libraries/ArduinoModbus/blob/master/src/libmodbus/modbus-rtu.cpp#L399

albydnc, 2018-10-22:
```
#include <ArduinoRS485.h>
#include <ArduinoModbus.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);
  if (!ModbusRTUClient.begin(9600)) {
    Serial.println("Failed to start Modbus RTU Client!");
    while (1);
  }
}

void loop() {

  if (!ModbusRTUClient.requestFrom(0x01, HOLDING_REGISTERS, 0x0016, 2)) {
    Serial.print("failed to read registers! ");
    Serial.println(ModbusRTUClient.lastError());
  }else{
    uint8_t word1 = ModbusRTUClient.read();
    uint8_t word2 = ModbusRTUClient.read();
    Serial.println(word1 << 16 | word2);
    
  }
  delay(5000);
}
```
That's the sketch I'm using, it is just a modified version of the example for a temperature sensor.

albydnc, 2018-10-22:
from moserial the response is 
00 01 03 04 00 00 00 45
after 5 sec 
3B C0 00 01 03
and it keeps going this way, it reads a 00 that should not be part of the message and then the CRC is shifted in the second read.

albydnc, 2018-10-22:
Hi @sandeepmistry, 
the hourly build  has solved the issue.
Thanks for your help.

closed 2018-10-22 by albydnc