arduino-libraries/ArduinoModbus #10: ModbusTCPClient has very long and unchangeable TimeOut

SokoFromNZ, 2019-07-19:
Hello,

I have an unreliable ModbusTCP device I want to read via this library which sometimes just doesn't reply.

It is unfortunate that the ModbusTCPClient waits 30 seconds for a reply. Even more bad: I cannot change this timeout (excerpt of ModbusTCPClient.cpp):

```
int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
{
  modbus_t* mb = modbus_new_tcp(_client, ip, port);

  // Here is the timeout set to 30 Seconds
  modbus_set_response_timeout(mb, 30, 0);

  return ModbusClient::begin(mb, MODBUS_TCP_SLAVE);
}
```

An overload would be great here! You can exchange the whole upper part with:

```
int ModbusTCPClient::begin(IPAddress ip, uint16_t port, uint32_t usecTimeout)
{
  modbus_t* mb = modbus_new_tcp(_client, ip, port);

  // Set Timeout to value provided
  modbus_set_response_timeout(mb, 0, usecTimeout);

  return ModbusClient::begin(mb, MODBUS_TCP_SLAVE);
}
int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
{
  // Keep standard timeout of 30 seconds
  return begin(ip, port, 30000000);
}
```

Do you think you can make this official so I can revert my local changes in your library.

Thanks
Soko

sandeepmistry, 2019-08-02:
Hi @SokoFromNZ,

What do you think about adding an `setTimeout` API, instead of adding a new `begin` API?

SokoFromNZ, 2019-08-04:
Hi @sandeepmistry 

Would be OK as well as long as it works. Meaning: Do I have to make sure to call the `setTimeout(..)` before the `begin(..)`? Or does a call to `setTimeout(..)` - respectively `modbus_set_response_timeout(..)` - during a connection affect the timeout? I haven't had a look into the latter method...

Soko

sandeepmistry, 2019-08-07:
Hi @SokoFromNZ,

I think we can do two things:

1) If you want to set the timeout for connection, it must be called before `setTimeout(...)` must be called before `begin(...)`.
2) If you want to adjust the response time out value after `begin(...)` it can call `modbus_set_response_timeout(...)`

Between 1) and 2), the `_mb` member var should be NULL in 1). Maybe it makes sense to put this in the `ModbusClient` class?

SokoFromNZ, 2019-08-08:
Sounds like a plan!

If the `modbus_set_response_timeout(...)` works the same for an RTU client the `setTimeout(..)` definitely belongs into the `ModbusClient` class.

I'm looking forward to your implementation of this and the next official version of this library.

sandeepmistry, 2019-08-08:
Hi @SokoFromNZ,

Do you have time to try to make a pull request for this?

SokoFromNZ, 2019-08-08:
Heya,

Its been a while since I've done anything with Github (using TFS usually) and I'm not quite sure how to do this. And I've only done it once... All I remember is that it was quite a hussle if you are not used to doing this on regular bases.

Can I create a pull request directly in the Github web interface? Or do I have to have Visual Studio (Code) installed?

I could give it a try but it may take a couple of weeks until I find time to play around with it. It might be just 5 minutes for you... thats why I thought it would be more efficient if you do it.

Do you have time for it?

Thanks
Soko

sandeepmistry, 2019-09-09:
Hi @SokoFromNZ,

I've made pull request #15 based on our discussion. Please let me know of your thoughts, you can download a zip file that includes the changes here: https://github.com/sandeepmistry/ArduinoModbus/archive/client-timeout.zip

SokoFromNZ, 2019-09-10:
Hi,

Thanks a lot for your effort and your time. I've had a look through all your changes. It does even more than I need and I haven't found any errors as well :)

Please let me know once you merged it into main (is this the correct phrasing for GitHub?).

Thanks again
Soko

sandeepmistry, 2019-09-10:
Hi @SokoFromNZ,

Thank you for you response. Just to confirm, you have downloaded and tried the code changes?

> Please let me know once you merged it into main (is this the correct phrasing for GitHub?).

Yes, merged is the correct term :)

You will automatically be informed once the changes are merged by Github, as the issue will be closed.

SokoFromNZ, 2019-09-11:
Heya,

No, I havent tried the code. I'm on vacation at the moment. I wont be able to test it on my Android system until in roughly 2 weeks time.

Soko

SokoFromNZ, 2019-09-30:
Hi again,

I've tested know your source from the ZIP file and everything works as expected. Please merge it to the master asap.

Thanks heaps!
Soko

sandeepmistry, 2019-09-30:
Hi @SokoFromNZ,

Thanks for testing and your feedback! I've just merged pull request #15 into master.

closed 2019-09-30 by sandeepmistry

tomtobback, 2020-02-12:
hi guys, probably an easy question but i'm not familiar with how classes work: how to call this setTimeout() function? i've started from the WifiModbusClientToggle.ino example, i modified it for Ethernet, it's running well but locking for 30 seconds from time to time so i want to adjust the timeout.
```
EthernetClient client_modbus;
ModbusTCPClient modbusTCPClient(client_modbus);
```
and in loop:
```
  if (!modbusTCPClient.connected()) {
    // client not connected, start the Modbus TCP client
    Serial.println("Attempting to connect to Modbus TCP server");
    if (!modbusTCPClient.begin(server)) {
```
i can't call `modbusTCPClient.setTimeout(1000);` as it is not a member of that class, and i don't have the ModbusClient class available? how to access that function? thanks in advance