arduino-libraries/ArduinoModbus #16: Problem with single coil functions

Nickyy-b, 2019-09-13:
Hello sandeepmistry,

I need your help with this problem, I've managed to master a good part of the Arduinomodbus library, and in C language I have a great knowledge, but I really can't get over this problem, in example (ModbusRTUClientKitchenSink), after the void loop you specified functions for write a single coil, read single register, etc.(the comments after the voids of each function)
I am trying to use ModbusRTUClient.coilread, and also .write, but it returns me -1 or some strange character, sometimes does not return anything, other functions I use normally, but these I can't. Thanks if you can help me. I am using an Arduino mega 2560 as client, and a pro mini as server, I do not think the problem is in one of them because so far the Arduinos have done several things I did with this library, I will attach the part of the code as image. I don't know if I'm making a stupid mistake, don't notice my english, it's not the best :)
[image]
                  I've done without client.read, I've used other functions but it didn't work

sandeepmistry, 2019-09-18:
Hi @Nickyy-b,

Could you please provide more insight on how things are wired/connected?

I also believe since the Pro Mini only has a single `Serial` instance, any sketches should avoid `Serial.println(...)` statements and let the Modbus library have full control of the Serial port.

Nickyy-b, 2019-09-25:
Thanks for replying, sorry for the delay in returning.
I connected the client (Mega) and the server (Pro mini) directly Rx to Rx and Tx to Tx, that programming I attached is from mega, and the Serial window is the answer from Pro mini (server) returns to mega. The programming of the server is modified only with a char that makes a led flash, just this, I'm not using serial.print, strangely only this function of reading a single coil is returning me -1, the others read / change multiple registers / Coils work normally, I can change to read less than 10, 3 for example, but ModbusRTUClient.coilread returns me -1, can you help me?

sandeepmistry, 2019-10-07:
@Nickyy-b please share the exact sketches you are using. Are you able to also use a logic analyzer to trace what is being sent and received on the wire?

Nickyy-b, 2019-10-16:
The codes !!
[Edited.zip](https://github.com/arduino-libraries/ArduinoModbus/files/3731983/Edited.zip)

What didn't catch was the function in "if serial == d", this function returns me -1. Do you think I can replace it with the requestFrom function or RTU.beginTransmission, without any problem?