arduino-libraries/ArduinoModbus #18: ModbusServer::writeInputRegisters not implemented?

jlucas2019, 2019-10-21:
Hello,

When trying to use the function ModbusServer::writeInputRegisters got the following error:

/tmp/ccmQwHO6.ltrans1.ltrans.o: In function `main':
<artificial>:(.text.startup+0x996): undefined reference to `ModbusServer::writeInputRegisters(int, unsigned int*, int)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

The function is documented under the official ArduinoModbus library site:
https://www.arduino.cc/en/ArduinoModbus/ModbusServerwriteInputRegisters

However, looking inside the code, the function seams to be defined in ModbusServer.h, but not implemented in ModbusServer.cpp.

Is there any specific reason for this or is just a lapse?

Thanks,
Jorge

sandeepmistry, 2019-11-14:
Hi @jlucas2019,

Thanks for bringing this to our attention!

> Is there any specific reason for this or is just a lapse?

This was just a lapse. I've opened pull request #20 to correct this. It would be great if you could try out the changes and provide your feedback.

terrasmart, 2020-01-23:
Hello. 

Is the WriteMultipleHoldingRegisters function n.16 (10H) implemented for the ModbusClient Class ?

I don't find it. 

I need to use a 10H function because I have a sensor which accept only functions 03 and 16.

closed 2023-10-06 by per1234