arduino-libraries/ArduinoModbus #26: Conflict with serial MKR 1010

vri-subsea-tech, 2020-02-24:
Hi,

I tried the ModBusRTUClient class with the RS485 shield for MKR 1010. I had no troubles to read holding registers when I was debugging with USB plug connected.
But when I removed the cable, I was unable to read Modbus and I have a long timeout when I try to read it.
I tried to not use Serial USB, but i encountered the same problem, when I unplug the cable I was unable to read the modbus. I also tried to supply my board on USB 5V without communication and without success.

Someone already encountered this issue?

Thanks in advance!

cheetor5923, 2020-03-04:
Ahh I see it's a SAMD21G based board. They're unusually picky about SerialUSB.. It's really hard to see what is going on truely without a debugger like a Atmel-ICE...
First off I'd look at lowering the super long 1 second timeout in the library so a much more sensible number... I chose 5ms, a sensible timeout for modbus operating at 56700bps. 1 second is way too long.

ModbusRTUClient.setTimeout(5); //large timeouts need to be avoided
RS485.setTimeout(5);
MODBUS_PORT.begin(57600);
ModbusRTUClient.begin(57600);

engeen-nl, 2020-08-18:
I had the same problem using the MKR 1310 with the RS485 shield when powering it trough the RS485 shield. Apparently the MKR 1310's BQ24195L needs to be configured in such a way to generate 5V on the 5V pin for the shield to work. I wasn't able to do this but the shield makes a beautiful 5V on the Vin pin. For now I jumper the Vin and 5V pins and communication keeps going without the USB connected.

I don't know why you cannot debug over USB though, I don't have a problem doing so. I do notice though a delay is needed after writing or reading registers. I'm using 4 ms. But that can be because of my nodes. Anyone else has the same experience?

To bad ModbusRTUClient.setTimeout is not available yet in the latest version in the Library. Instead I adjusted ModbusRTUClient.cpp:
`modbus_set_response_timeout(mb, 0, 5000);`

Be careful, the third parameter of `modbus_set_response_timeout` is **micro**seconds. So 5000 means 5 milliseconds.

MachineSaver, 2020-11-30:
@vri-sst I know this is probably not the place to request help but I see from your response that you've been using the MKR1010 and an MKR485 shield and successfully reading/writing the holding registers. I am trying to do the same using the Arduino as a Modbus Server. I have used the "kitchen sink" example but I cannot read the holding registers that are setup in the example from the Arduino. I have no issue reading coils when using the "LED" example which sets up coils. Any help or pointing in the right direction would be much appreciated.

mikdur, 2021-05-16:
This same issue appears with the MKR1400 when powering it through the MKR RS485 shield. Following @engeen-nl 's suggestion to put a jumper between 5V and Vin is an interim workaround. However, when powering the modules through the attached LiPo accumulator, the RS485 shield will not respond either. 

The only two cases I identified where RS485 works as expected is when the MKR1400 is powered through USB or when it is powered through the screw connector on the RS485 shield with 5V and Vin connected to each other.