arduino-libraries/ArduinoModbus #29: Any chance to implement software serial? and several more.

ijash, 2020-03-22:
I've been reading [this good article](https://www.stderr.nl/Blog/Hardware/Electronics/Arduino/ModbusLibrary.html). He's been testing some libraries across the web and make some good comparison point.
let me quote it some for this particular library:

> https://github.com/arduino-libraries/ArduinoModbus
> - Official library from Arduino.
> - Master and slave.
> - Uses the RS485 library to communicate, but does not offer any way to pass a custom RS485 instance, so it is effectively hardcoded to a specific serial port.
> - Offers only single value reads and writes.
> - Slave stores value internally and reads/writes directly from those, without any callback or way to detect that communication has happened.

more on the page, and I hope this reference makes a good study for further improvement.
so, for my personal request, Any chance to implement to softwareSerial?
Because i want the official library can cover it all.

matthijskooijman, 2020-03-22:
What would be needed on this library is to allow passing a custom RS485 instance, rather than hardcoding to the predefined global RS485 instance.

Additionally, the RS485 library should also be modified to accept a `Stream` instead of [a `HardwareSerial` instance](https://github.com/arduino-libraries/ArduinoRS485/blob/49a349613c34f84fffd4fdd40849aefddb613e52/src/RS485.h#L41), so you can also pass SoftwareSerial (or other streams) instead of just HardwareSerial.

larsvankleef, 2021-06-22:
Would be pretty nice if this would be an option. Or is there another workaround to get this library working on an Arduino UNO