arduino-libraries/ArduinoModbus #35: Cannot run example program on uno wifi rev 2

fhainz, 2020-09-18:
Hello!

I tried to run the example program on an Arduino Uno Wifi Rev. 2 but i get errors on compiling.

<code>
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:27:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:35:25: error: expected ')' before '&' token
   ModbusTCPClient(Client& client);
                         ^
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:46:13: error: reference to 'IPAddress' is ambiguous
   int begin(IPAddress ip, uint16_t port = 502);
             ^~~~~~~~~
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:24:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.h:13:7: note: candidates are: class IPAddress
 class IPAddress;
       ^~~~~~~~~
In file included from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/../Client.h:23:0,
                 from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/Client.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:27:
/Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/../IPAddress.h:35:7: note:                 class arduino::IPAddress
 class IPAddress : public Printable {
       ^~~~~~~~~
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:27:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:46:13: error: 'IPAddress' has not been declared
   int begin(IPAddress ip, uint16_t port = 502);
             ^~~~~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:61:3: error: reference to 'Client' is ambiguous
   Client* _client;
   ^~~~~~
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:24:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.h:12:7: note: candidates are: class Client
 class Client;
       ^~~~~~
In file included from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/Client.h:23:0,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:27:
/Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/../Client.h:27:7: note:                 class arduino::Client
 class Client : public Stream {
       ^~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:29:33: error: expected constructor, destructor, or type conversion before '(' token
 ModbusTCPClient::ModbusTCPClient(Client& client) :
                                 ^
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:38:28: error: 'int ModbusTCPClient::begin' is not a static data member of 'class ModbusTCPClient'
 int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
                            ^~~~~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:38:28: error: reference to 'IPAddress' is ambiguous
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:24:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.h:13:7: note: candidates are: class IPAddress
 class IPAddress;
       ^~~~~~~~~
In file included from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/../Client.h:23:0,
                 from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/Client.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:27:
/Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/deprecated/../IPAddress.h:35:7: note:                 class arduino::IPAddress
 class IPAddress : public Printable {
       ^~~~~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:38:51: error: expected primary-expression before 'port'
 int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
                                                   ^~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp: In member function 'int ModbusTCPClient::connected()':
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:49:10: error: '_client' was not declared in this scope
   return _client->connected();
          ^~~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/ModbusTCPClient.cpp:49:10: note: suggested alternative: 'Client'
   return _client->connected();
          ^~~~~~~
          Client
*** [.pio/build/uno_wifi_rev2/libe31/ArduinoModbus/ModbusTCPClient.cpp.o] Error 1
[... 192 lines cut ...]
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:916:34: note: suggested alternative: 'Client'
 modbus_t* modbus_new_tcp(Client* client, IPAddress ip_address, int port)
                                  ^~~~~~
                                  Client
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:916:42: error: reference to 'IPAddress' is ambiguous
 modbus_t* modbus_new_tcp(Client* client, IPAddress ip_address, int port)
                                          ^~~~~~~~~
In file included from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:81:0:
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.h:13:7: note: candidates are: class IPAddress
 class IPAddress;
       ^~~~~~~~~
In file included from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/Client.h:23:0,
                 from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/Arduino.h:23,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp-private.h:12,
                 from .pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:82:
/Users/fabian/.platformio/packages/framework-arduino-megaavr/cores/arduino/api/IPAddress.h:35:7: note:                 class arduino::IPAddress
 class IPAddress : public Printable {
       ^~~~~~~~~
.pio/libdeps/uno_wifi_rev2/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:916:64: error: expected primary-expression before 'int'
 modbus_t* modbus_new_tcp(Client* client, IPAddress ip_address, int port)
                                                                ^~~
*** [.pio/build/uno_wifi_rev2/libe31/ArduinoModbus/libmodbus/modbus-tcp.cpp.o] Error 1</code>

Is the uno wifi rev 2 not supported?

cdh4u, 2020-09-18:
I just tried to run the example with an uno wifi rev2 too, and got the same error.

KLVN, 2020-11-19:
I have the same problem. ArduinoModbus was running for nearly one year straight on my Arduino Uni Wifi Rev2 and I just wanted to make some small changes to my code and updated some libraries (probably I also have a later version of Arduino IDE nowadays). Since then I can't compile any file if `#include <ArduinoModbus.h>` is included. Not even the basic Modbus examples are working...

```c
In file included from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:27:0:
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:35:25: error: expected ')' before '&' token
   ModbusTCPClient(Client& client);
                         ^
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:46:13: error: reference to 'IPAddress' is ambiguous
   int begin(IPAddress ip, uint16_t port = 502);
             ^~~~~~~~~
In file included from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:24:0:
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\libmodbus/modbus-tcp.h:13:7: note: candidates are: class IPAddress
 class IPAddress;
       ^~~~~~~~~
In file included from c:\users\kelvin\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\client.h:23:0,
                 from C:\Users\Kelvin\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Client.h:23,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:23,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:27:
c:\users\kelvin\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\IPAddress.h:35:7: note:                 class arduino::IPAddress
 class IPAddress : public Printable {
       ^~~~~~~~~
In file included from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:27:0:
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:46:13: error: 'IPAddress' has not been declared
   int begin(IPAddress ip, uint16_t port = 502);
             ^~~~~~~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:61:3: error: reference to 'Client' is ambiguous
   Client* _client;
   ^~~~~~
In file included from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:24:0:
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\libmodbus/modbus-tcp.h:12:7: note: candidates are: class Client
 class Client;
       ^~~~~~
In file included from C:\Users\Kelvin\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Client.h:23:0,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:23,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:27:
c:\users\kelvin\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\client.h:27:7: note:                 class arduino::Client
 class Client : public Stream {
       ^~~~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:29:33: error: expected constructor, destructor, or type conversion before '(' token
 ModbusTCPClient::ModbusTCPClient(Client& client) :
                                 ^
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:38:28: error: 'int ModbusTCPClient::begin' is not a static data member of 'class ModbusTCPClient'
 int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
                            ^~~~~~~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:38:28: error: reference to 'IPAddress' is ambiguous
In file included from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:24:0:
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\libmodbus/modbus-tcp.h:13:7: note: candidates are: class IPAddress
 class IPAddress;
       ^~~~~~~~~
In file included from c:\users\kelvin\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\client.h:23:0,
                 from C:\Users\Kelvin\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Client.h:23,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.h:23,
                 from C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:27:
c:\users\kelvin\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\IPAddress.h:35:7: note:                 class arduino::IPAddress
 class IPAddress : public Printable {
       ^~~~~~~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:38:51: error: expected primary-expression before 'port'
 int ModbusTCPClient::begin(IPAddress ip, uint16_t port)
                                                   ^~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp: In member function 'int ModbusTCPClient::connected()':
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:49:10: error: '_client' was not declared in this scope
   return _client->connected();
          ^~~~~~~
[... cut ...]
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp: In member function 'virtual void ModbusTCPServer::poll()':
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp:62:7: error: '_client' was not declared in this scope
   if (_client != NULL) {
       ^~~~~~~
C:\Users\Kelvin\Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp:62:7: note: suggested alternative: 'Client'
   if (_client != NULL) {
       ^~~~~~~
       Client
```

facchinm, 2020-11-20:
@fhainz @KLVN @cdh4u I just pushed a fix for Uno WiFI rev2 (and all cares derived by ArduinoCore-API in fact).
This is the commit https://github.com/arduino-libraries/ArduinoModbus/commit/1ef344f194d51a8cfddd35d8fd8adb575c6f12b3 and it is contained in v1.0.2 (you should get it via library manager in a few hours).
Closing the issue, please reopen if it's not really fixed or if there are other related problems

closed 2020-11-20 by facchinm