arduino-libraries/ArduinoModbus #38: Compile Error

chrisflesher, 2020-11-13:
/home/chris/Arduino/libraries/ArduinoModbus/src/libmodbus/modbus-tcp.cpp:32:21: error: expected unqualified-id before '{' token
   32 | #define printf(...) {}
      |                     ^
/home/chris/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/Print.h:88:10: note: in expansion of macro 'printf'
   88 |     void printf(const char[], ...);
      |          ^~~~~~

chrisflesher, 2020-11-13:
I tried adding a `#define DEBUG` before `#include <ArduinoModbus.h` and it didn't fix it but commenting out this block in `modbus-tcp.cpp` got rid of the compile error:
```
// #ifndef DEBUG
// #define printf(...) {}
// #define fprintf(...) {}
// #endif
```

tgthomas, 2020-11-16:
Yes, compiling the RTU examples for Adafruit M0 and M4 produces exactly the same error, but not so for an Arduino AVR micro.
Commenting out (just) the #define printf(...) { } in modbus-tcp.cpp (line 32) avoids the compile error.