arduino-libraries/ArduinoModbus #39: Cannot run example program on uno wifi rev 2

fld113, 2020-11-28:
Despite the last fix, I cannot run the exemple program on uno wifi rev 2. I used the code in the ArduinoModbus version 1.02 modified 8 days ago, and the IDE 1.8.13 under OSX. I get the following message : 
Arduino : 1.8.13 (Mac OS X), Carte : "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

/Users/xxx/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/main.cpp: In function 'main':
/Users/xxx/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Stream.h:50:7: warning: 'MEM[(struct Stream &)&client]._startMillis' may be used uninitialized in this function [-Wmaybe-uninitialized]
 class Stream : public Print
       ^
/Users/xxx/Documents/Arduino/libraries/WiFiNINA/src/WiFiServer.cpp:69:20: note: 'MEM[(struct Stream &)&client]._startMillis' was declared here
         WiFiClient client(sock);
                    ^
/Users/xxx/Documents/Arduino/libraries/WiFiNINA/src/WiFiClient.h:28:7: warning: 'client._socket' may be used uninitialized in this function [-Wmaybe-uninitialized]
 class WiFiClient : public Client {
       ^
/Users/xxx/Documents/Arduino/libraries/WiFiNINA/src/WiFiServer.cpp:69:20: note: 'client._socket' was declared here
         WiFiClient client(sock);
                    ^
/var/folders/kf/5_pjskbd1yg3j6qlthc39v5m0000gn/T//ccpthV3p.ltrans0.ltrans.o: In function `begin':
/Users/xxx/Documents/Arduino/libraries/ArduinoModbus-master/src/ModbusTCPServer.cpp:40: undefined reference to `modbus_new_tcp'
/Users/xxx/Documents/Arduino/libraries/ArduinoModbus-master/src/ModbusTCPServer.cpp:46: undefined reference to `modbus_tcp_listen'
/var/folders/kf/5_pjskbd1yg3j6qlthc39v5m0000gn/T//ccpthV3p.ltrans0.ltrans.o: In function `accept':
/Users/xxx/Documents/Arduino/libraries/ArduinoModbus-master/src/ModbusTCPServer.cpp:55: undefined reference to `modbus_tcp_accept'
collect2: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Arduino Uno WiFi Rev2

Did I missed something ?

Best Regards
Frederic

sunn78, 2020-12-25:
Hi,

May be the problem is the last actualization of mkr samd21 board, try to came back and compile again.

johnfing, 2021-01-03:
I just tried compiling the sample program against the latest source (1/3/2021) with the uno wifi rev2, and I get the following:
C:\Users\parents\AppData\Local\Temp\ccl2x1Ra.ltrans0.ltrans.o: In function `begin':
C:\Users\parents\Documents\Arduino\libraries\ArduinoModbus\src/ModbusTCPServer.cpp:40: undefined reference to `modbus_new_tcp'
C:\Users\parents\Documents\Arduino\libraries\ArduinoModbus\src/ModbusTCPServer.cpp:46: undefined reference to `modbus_tcp_listen'
C:\Users\parents\AppData\Local\Temp\ccl2x1Ra.ltrans0.ltrans.o: In function `accept':
C:\Users\parents\Documents\Arduino\libraries\ArduinoModbus\src/ModbusTCPServer.cpp:55: undefined reference to `modbus_tcp_accept'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

sunn78, 2021-01-04:
Hi,

You must install 1.8.9 instead 1.8.10 / 11

facchinm, 2021-01-04:
The issue should now be fixed in version 1.0.3 of the library on every core/board combination (without any need to revert)

closed 2021-01-04 by facchinm