arduino-libraries/ArduinoModbus #46: holdingRegisterWrite() stop after some hours

sunn78, 2021-01-06:
Hi,

I have the example led which I've added holdingRegisterWrite function for lm35 sensor temperature.

All works fine, but after some hours holding Register stops working, coil register keeps working properly.

Any ideas?

sunn78, 2021-01-14:
Hi,
 I've tested the basic example and it stops after some hours too.

It's my mkr broken or the library doesn't work properly?

I've seen that mkr it's connected on the router, so the problem looks about modbus part only.

Aquadukt, 2021-01-23:
Seems to be the same Problem like described here:
https://github.com/arduino-libraries/ArduinoModbus/issues/43#issuecomment-752368748

I don't find a sufficient solution. Only some workarrounds by reducing the polling frequency. But this communication lib seems still be buggy. Unfortunately there seems no attention or feedback from original developers to this tpoic.

sunn78, 2021-01-26:
Hi,

https://github.com/sfera-labs/iono/blob/master/IonoModbusTcp/examples/IonoModbusTcp/IonoModbusTcp.ino

Here there are some interesting things, like inside loop, but I haven't got knowledge to aplicate it in our basic sketch.