arduino-libraries/ArduinoModbus #48: Problems debugging the library

SWoto, 2021-01-26:
Hey everyone,
hope you're all safe and sound (as possible) during these dark times.

## Problem
I'm using the ArduinoModbus library and so far so good. However, yesterday, I tried communicating with a diferent hardware with the Arduino as the slave to have its input registers read. The code, Modbus RTU Server Kitchen Sink with little modifications, receives the message right (added print function to the .cpp) but doesn't reply as it should.

Due to most past of the core code being in `c` i can't use `Serial.print()` to debug and looks like `printf` and `fprintf` doesn't work (at least not for me).

Have anyone managed to make those two functions work? How are you debugging it?

per1234, 2021-01-26:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

SWoto, 2021-01-26:
@per1234 

Probably due to the hardware (SAMD), the code tries to reply too early and it isn't able to do so. I checked with the oscilloscope and although the debugging showed that the reply was correctly built and sent, the signal wasn't there.

I've tried adding some delay to check if this was the problem and it solved. I tested from 5ms to 500ms and if worked form 110 to 500.

**Before**
```Cpp
void ModbusRTUServerClass::poll()
{
  uint8_t request[MODBUS_RTU_MAX_ADU_LENGTH];

  int requestLength = modbus_receive(_mb, request);

  if (requestLength > 0) {
    modbus_reply(_mb, request, requestLength, &_mbMapping);
  }
}

```

**After**
```Cpp
void ModbusRTUServerClass::poll()
{
  uint8_t request[MODBUS_RTU_MAX_ADU_LENGTH];

  int requestLength = modbus_receive(_mb, request);

#ifdef ARDUINO_ARCH_SAMD
  delay(110);
#endif
  modbus_reply(_mb, request, requestLength, &_mbMapping);
  }
}
```