arduino-libraries/ArduinoModbus #51: ModbusRTUClient.read()

IanKPC, 2021-03-27:
Hopefully someone can point me in the right direction, im probably making this more complicated than it really is.

When a requestform is sent to the slave node, and the client gets a response. Where is that response stored in memory?

per1234, 2021-03-27:
Hi @IanKPC. Thanks for your interest in the ArduinoModbus library. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2021-03-27 by per1234

IanKPC, 2021-03-27:
My apologies, thank you for the quick response.

Have a good day!