arduino-libraries/ArduinoModbus #64: Crash after 1.0.4 update

jeromegirard, 2021-06-02:
Hi,

I just update to 1.0.4 and my code crash (reboot by WDT) on the following line
`...
	if (!ModbusRTUClient.begin(9600)) {
...`

Arduino MKRZero + MKR RS485 shield + MKR Ethernet shield

Rollback to 1.0.3 and everything is OK :)

legicore, 2021-06-02:
Hi,

I can confirm this - I'm also using the **Arduino MKR ZERO** with the **Arduino MKR RS485 Shield**.
My code got stuck when calling `int ModbusRTUClientClass::begin(unsigned long baudrate, uint16_t config)`.
It's working normal again after rollback to 1.0.3.

After some rough (Serial.print()) debugging, it looks like it stuck while calling `int modbus_connect(modbus_t *ctx)` in `int ModbusClient::begin(modbus_t* mb, int defaultId)` (_ModbusClient.cpp:58_).

facchinm, 2021-06-07:
Hi,
the issue was a missing default value (it was ok in ModbusRTUServer :disappointed: ), sorry for the inconvenience. I also pushed version 1.0.5 that should be available in the next few hours via the library manager

closed 2021-06-07 by facchinm