arduino-libraries/ArduinoModbus #65: Compiler Warnings

PaulStoffregen, 2021-06-09:
This library has many compiler warnings.  For example, compiling for SAMD MKR1000 with Preferences set to All warnings gives this:

```
In file included from /home/paul/teensy/sketch/libraries/ArduinoModbus/src/ModbusRTUClient.cpp:27:0:
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/ModbusRTUClient.h: In constructor 'ModbusRTUClientClass::ModbusRTUClientClass(RS485Class&)':
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/ModbusRTUClient.h:44:25: warning: 'ModbusRTUClientClass::_rs485' will be initialized after [-Wreorder]
   RS485Class* _rs485 = &RS485;
                         ^~~~~
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/ModbusRTUClient.cpp:35:36: warning:   base 'ModbusClient' [-Wreorder]
   _rs485(&rs485), ModbusClient(1000)
                                    ^
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/ModbusRTUClient.cpp:34:1: warning:   when initialized here [-Wreorder]
 ModbusRTUClientClass::ModbusRTUClientClass(RS485Class& rs485) :
 ^~~~~~~~~~~~~~~~~~~~
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c: In function '_sleep_response_timeout':
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c:161:5: warning: implicit declaration of function 'delay' [-Wimplicit-function-declaration]
     delay(ctx->response_timeout.tv_sec * 1000);
     ^~~~~
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c:162:5: warning: implicit declaration of function 'delayMicroseconds' [-Wimplicit-function-declaration]
     delayMicroseconds(ctx->response_timeout.tv_usec);
     ^~~~~~~~~~~~~~~~~
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c: In function '_modbus_receive_msg':
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c:501:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
                 if (length_to_read != 0) {
                    ^
/home/paul/teensy/sketch/libraries/ArduinoModbus/src/libmodbus/modbus.c:505:13: note: here
             case _STEP_META:
             ^~~~
```

The "implicit declaration of function" warnings are only because Arduino.h is not included, by code using Arduino APIs.  Easy to fix.

facchinm, 2021-06-22:
Fixed by https://github.com/arduino-libraries/ArduinoModbus/commit/b41428b20a05c60a371e95e269426f6c779e26bf, thanks for pointing this out :wink:

closed 2021-06-22 by facchinm

PaulStoffregen, 2021-06-22:
Please also include Arduino.h in modbus.c.

The compiler warning about implicit declaration of delayMicroseconds() becomes a compiler error if the core library implements it as an inline function.

Pull request here
https://github.com/arduino-libraries/ArduinoModbus/pull/66/files