arduino-libraries/ArduinoModbus #96: Multiple connected clients

salasidis, 2022-07-13:
The usual code for servicing a clients only works with one connected client at a time.

`while (1) {
    
    ethClient = ethServer.available();
    
    if (ethClient) {

        modbusTCPServer.accept(ethClient);

        while (ethClient.connected()) {
          modbusTCPServer.poll();
`

Is there a correct way (or an example) for how to connect multiple clients to the server at one time.

Thanks

khoih-prog, 2022-10-01:
Hi @salasidis 

You can try this Async [**eModbus** library](https://github.com/eModbus/eModbus) I just found out. It's designed for **ESP32**, but IMO, it's easy to port to other platforms, such as `mbed_portenta`, `ESP8266`, `arduino-pico`, , `STM32`, etc.

From https://emodbus.github.io/

> This is a library to provide Modbus client (formerly known as master), server (formerly slave) and bridge/gateway functionalities for Modbus RTU, ASCII and TCP protocols.
> 
> Modbus communication is done in separate tasks, so Modbus requests and responses are non-blocking. Callbacks are provided to prepare or receive the responses asynchronously.
> 
> There is a synchronous interface available as well where requests are waiting for their responses to arrive.

I don't have any Modbus experience or equipment, If you'd like to try and are still interested, I can port it to multiple platforms. But you have to do all the tests.

salasidis, 2022-10-01:
I do have a license to modbus Poll, and could buy a license for modbus server if required  (from www.modbustools.com).

this is what I use to test my current project's modbus server connection. It would be willing to rewrite my code to add the use of this library to see if I can have multiple poll clients from different sites accessing the software.

I don't have any prewritten software to test the client interface, but presumably we could use the examples o test for that.

codev123, 2023-07-20:
hi @khoih-prog , can you please port emodbus library to rp2040 arduino pico core. i will try to run tests. thank you sir

codev123, 2023-10-09:
> Hi @salasidis
>
> You can try this Async [**eModbus** library](https://github.com/eModbus/eModbus) I just found out. It's designed for **ESP32**, but IMO, it's easy to port to other platforms, such as [...]

sir, i can run the tests if you port the library

codev123, 2023-10-13:
hi @salasidis , how did you connect multiple clients to connect to tcp server, please help if if were able to do that.