arduino-libraries/ArduinoModbus #99: Change Serial port?

RobWPink, 2022-09-05:
Is there a way to change serial port for RTU comms to Serial1 so that I can do two simultaneous rtu loops (one slave one master) on half duplex?

per1234, 2022-09-05:
Hi @RobWPink. There is a description of the work that will be required on the library in order to add this capability here:

https://github.com/arduino-libraries/ArduinoModbus/issues/29#issuecomment-602181613

Even though this issue overlaps with https://github.com/arduino-libraries/ArduinoModbus/issues/29, it is somewhat different in that the software serial support requested in https://github.com/arduino-libraries/ArduinoModbus/issues/29 will also require a change in the ArduinoRS485 library (https://github.com/arduino-libraries/ArduinoRS485/issues/9), so I think we can consider your request to be distinct from https://github.com/arduino-libraries/ArduinoModbus/issues/29 rather than a duplicate.

pinpong, 2022-11-05:
You can change to another serial port with:

```c
#include <ArduinoModbus.h>

RS485Class _RS485(Serial1, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
ModbusRTUServer.begin(_RS485, 42, 9600);
```

In my case i wasted 1 day to check that the Arduino Nano Every is using Serial1.

Breakaway007, 2022-11-09:
> 

Hi, I am also struggling with this, can you please elaborate a bit on what you did to solve this?

rozrabiak, 2023-07-25:
With STMDuino not working, I'm using BlackPill with STM32F4CCU6
Error:

`.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.cpp:209:104: error: no matching function for call to 'RS485Class::RS485Class(USBSerial&, int, <unnamed enum>, <unnamed enum>)'
  209 | RS485Class RS485(SERIAL_PORT_HARDWARE, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
      |                                                                                                        ^
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.cpp:34:1: note: candidate: 'RS485Class::RS485Class(HardwareSerial&, int, int, int)'
   34 | RS485Class::RS485Class(HardwareSerial& hwSerial, int txPin, int dePin, int rePin) :
      | ^~~~~~~~~~
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.cpp:34:40: note:   no known conversion for argument 1 from 'USBSerial' to 'HardwareSerial&'
   34 | RS485Class::RS485Class(HardwareSerial& hwSerial, int txPin, int dePin, int rePin) :
      |                        ~~~~~~~~~~~~~~~~^~~~~~~~
In file included from .pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.cpp:20:
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.h:53:7: note: candidate: 'constexpr RS485Class::RS485Class(const RS485Class&)'
   53 | class RS485Class : public Stream {
      |       ^~~~~~~~~~
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.h:53:7: note:   candidate expects 1 argument, 4 provided
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.h:53:7: note: candidate: 'constexpr RS485Class::RS485Class(RS485Class&&)'
.pio\libdeps\blackpill_f401cc\ArduinoRS485\src\RS485.h:53:7: note:   candidate expects 1 argument, 4 provided
*** [.pio\build\blackpill_f401cc\lib961\ArduinoRS485\RS485.cpp.o] Error 1`

> You can change to another serial port with:
> 
> ```c
> #include <ArduinoModbus.h>
> 
> RS485Class _RS485(Serial1, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
> ModbusRTUServer.begin(_RS485, 42, 9600);
> ```
> 
> In my case i wasted 1 day to check that the Arduino Nano Every is using Serial1.

steppen7, 2023-12-27:
> You can change to another serial port with:
> 
> ```c
> #include <ArduinoModbus.h>
> 
> RS485Class _RS485(Serial1, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
> ModbusRTUServer.begin(_RS485, 42, 9600);
> ```
> 
> In my case i wasted 1 day to check that the Arduino Nano Every is using Serial1.

Did not work for me using ESP32 devkitC.
```
...Arduino\libraries\ArduinoRS485\src\RS485.cpp:209:18: error: 'SERIAL_PORT_HARDWARE' was not declared in this scope
 RS485Class RS485(SERIAL_PORT_HARDWARE, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
                  ^~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

Tried declaring below in my code but it made no difference, probably something deeper down needs to be rewritten or defined for this to work with ESP32?
`#define SERIAL_PORT_HARDWARE Serial2`