arduino-libraries/ArduinoModbus #102: Possible to add multiple blocks of holding registers?

nikstp, 2022-10-17:
Hi, 

I'm trying to have an Arduino Uno serve as a Modbus RTU server to simulate a device I need for work. I'd like to setup multiple holding register blocks but so far I'm not having any luck as I get the "illegal data address" error. 

What I mean is that if I would like to read from holding registers 6, 254, 867, and 1023 then I can't get it to work just by running 

```
   //Configure 1024 registers
  result = ModbusRTUServer.configureHoldingRegisters(0, 1023) ;
  if(result != 0)
  {
      Serial.println("Holding registers configured incorrectly!");
  }

//initialize the 1024 registers with the number of the register
for(int i = 0; i < 1024; i++)
{
    result =    result = ModbusRTUServer.holdingRegisterWrite(i, (uint16_t) i);
        if(result != 1)
        {
            Serial.println("Failed to write to the holding registers!");
          break;
        }
}
```

What would be the correct way to go about doing this? 

I know this was just opened and closed, but I've used a wrong account for that. 

Thank you!

BayesianScrub, 2022-12-14:
`result = result = ModbusRTUServer.holdingRegisterWrite(i, (uint16_t)i);`
Doesn't look right

Also check how much ram you are using, the Arduino Uno Rev3 only has 2kB and you are trying to allocate 2kB for Modbus registers on top of whatever else is going on.

nikstp, 2022-12-14:
@BayesianScrub 

> result = result = ModbusRTUServer.holdingRegisterWrite(i, (uint16_t)i);
>Doesn't look right

That was a typo.

>Also check how much ram you are using, the Arduino Uno Rev3 only has 2kB and you are trying to allocate 2kB for Modbus >registers on top of whatever else is going on.

Possible cause, but if I remember correctly I've tried with 100+ and 200+ as well and what worked consistently is only 8 consecutive registers. Maybe it's more to do with how Modbus is defined in the lib; it's been a while so I'd have to check again on the number that works and if it's possible to define those registers separately. My recollection is that did not work, but maybe I didn't test it right.