arduino-libraries/ArduinoModbus #126: Not able to read Holding registers more than 15. RX Buffer issue?

codev123, 2023-10-11:
hello sir, i am using arduino modbus library with rp2040 pico pi with arduino pico core, i am able to communicate with modbus slave and client using this library, the limitation i am facing is that i need to read around 100 holding registers from energy meter slave and process that data to serve it on to modbus tcp server. but i can only read upto 14 holding registers using modbus rtu client. please guide me how to fix the issue.

const int numHoldingRegisters = 100;
void readHoldingRegisterValues() {
  Serial.print("Reading Holding Register values ... ");

  // read 10 Input Register values from (slave) id 42, address 0x00
  if (!ModbusRTUClient.requestFrom(42, HOLDING_REGISTERS, 0x00, numHoldingRegisters)) {
    Serial.print("failed! ");
    Serial.println(ModbusRTUClient.lastError());
  } else {
    Serial.println("success");

    while (ModbusRTUClient.available()) {
      Serial.print(ModbusRTUClient.read());
      Serial.print(' ');
    }
    Serial.println();
  }

  // Alternatively, to read a single Holding Register value use:
  // ModbusRTUClient.holdingRegisterRead(...)
}
the error shows ..............   failed! Connection timed out

pomplesiegel, 2023-12-18:
@codev123, any update / status + additional info? Thanks

codev123, 2023-12-19:
I have increased serial buffer size for uart to 256 bytes and issue is resolved. Thankyou sir

pomplesiegel, 2023-12-19:
@codev123, thanks for the update! Can you specify which specific file / line you changed for this? Thank you!

AlexanderTonn, 2023-12-19:
Please consider that Modbus RTU requires a termination resistor at the end of the line. i personally don't have problems with the buffer size. I used the Arduino Opta RS485 and i read currently 30 registers without problems. 

**Probably it can be helpful:** 
[I currently writing a class for reading and writing registers of a mppt charger. ](https://github.com/AlexanderTonn/SolarPowerhouse/blob/main/Code/SolarPowerhouse/modbusRTU_toyo.ino)

codev123, 2023-12-19:
> @codev123, thanks for the update! Can you specify which specific file / line you changed for this? Thank you!

I have used serial1 on my pico board, added thin line in setup() code
Serial1.setFIFOSize(256);
And it works fine.
Issue with library is that it doesn't fix modbus response buffer size, it's microcontroller specific.