arduino-libraries/ArduinoModbus #139: Programming issue? timeout without delay within the code

realKamillo, 2024-01-31:
Hello,

please find my example below based on Kitchen Sink example.
If I run the code with delay(5) at the beginning and at the end of the loop, everything is working fine, without I am running into timeout issue during "readInputRegisterValues();

Any idea what is wrong with my code or is this a bug?

Slave settings:
Baud rate: 9600
Parity bits: None

`/*
  Modbus RTU Client Kitchen Sink

  This sketch creates a Modbus RTU Client and demonstrates
  how to use various Modbus Client APIs.

  Circuit:
   - MKR board
   - MKR 485 shield
     - ISO GND connected to GND of the Modbus RTU server
     - Y connected to A/Y of the Modbus RTU server
     - Z connected to B/Z of the Modbus RTU server
     - Jumper positions
       - FULL set to OFF
       - Z \/\/ Y set to ON

  created 18 July 2018
  by Sandeep Mistry
*/

#include <ArduinoRS485.h> // ArduinoModbus depends on the ArduinoRS485 library
#include <ArduinoModbus.h>

int counter = 0;
byte readModbus1 = 0; 
byte writeModbus1 = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("Modbus RTU Client Kitchen Sink");

  // start the Modbus RTU client
  if (!ModbusRTUClient.begin(9600, SERIAL_8N1)) {
    Serial.println("Failed to start Modbus RTU Client!");
    while (1);
  }
}

void loop() {
  readInputRegisterValues();  

  counter++;
  //delay(5);

  if ((bitRead(readModbus1,0)==0) && (bitRead(readModbus1,1)==0) && (bitRead(readModbus1,2)==0) && (bitRead(readModbus1,3)==0) && (bitRead(readModbus1,4)==0) && (bitRead(readModbus1,5)==0) && (bitRead(readModbus1,6)==0) && (bitRead(readModbus1,7)==0))
    {
    ModbusRTUClient.beginTransmission(01, COILS, 0x00, 8);
    for (int i = 0; i < 8; i++) {
      ModbusRTUClient.write(0);
    }
    if (!ModbusRTUClient.endTransmission()) {
      Serial.print(F("failed!1 "));
      Serial.println(ModbusRTUClient.lastError());
    } else {
      Serial.println(F("success1"));
    }
  }
  if (bitRead(readModbus1,0)==1) {
    Serial.println(F("Bit3 ist gesetzt"));
  }

  if ((bitRead(readModbus1,0)==1) || (bitRead(readModbus1,1)==1) || (bitRead(readModbus1,2)==1) || (bitRead(readModbus1,3)==1) || (bitRead(readModbus1,4)==1) || (bitRead(readModbus1,5)==1) || (bitRead(readModbus1,6)==1) || (bitRead(readModbus1,7)==1)) 
  {
 
      writeModbus1 = bitWrite(writeModbus1,0, 1); //Update of relevant bits
      writeModbus1 = bitWrite(writeModbus1,2, 1); //Update of relevant bits
      writeModbus1 = bitWrite(writeModbus1,7, 1); //Update of relevant bits
      ModbusRTUClient.beginTransmission(01, COILS, 0x00, 8); //Start Modbus communication 

      for (int i = 0; i < 8; i++) {
        ModbusRTUClient.write(bitRead(writeModbus1,i)); //writes full byte on the Modbus
      }

    if (!ModbusRTUClient.endTransmission()) {
      Serial.print(F("failed!2 "));
      Serial.println(ModbusRTUClient.lastError());
    } else {
      Serial.println(F("success2"));
    }
  }
  //delay(5);
  //Serial.println();
}

void readInputRegisterValues() {
  Serial.print("Reading input register values ... ");

  // read 10 discrete input values from (slave) id 42,
  if (!ModbusRTUClient.requestFrom(1, DISCRETE_INPUTS, 0x00, 8)) {
    Serial.print("failed!3 ");
    Serial.println(ModbusRTUClient.lastError());
  } else {
    Serial.println("success3");

    while (ModbusRTUClient.available()) {
      readModbus1 = ModbusRTUClient.read();
      Serial.print(readModbus1);
    }
    Serial.println();
  }
}`

PanzerFowst, 2024-02-13:
What are you communicating with?  Check this forum post for an explanation on getting Modbus between two Arduino boards: https://forum.arduino.cc/t/arduinomodbus-only-prints-fail-messages-using-mkr1010-board-and-mkr485-shield/1197701.

realKamillo, 2024-02-14:
Hi,
I used a teensy 4.1 as Master and SGI0B08 (8DI 8DO by eletechsup). This is a simple Digital Input and Output card.
Strange is, that everything is working fine in case delay(5) is within the code. 

I will check the link you posted and scan for hints which could solve my problem. Thank you.