arduino-libraries/ArduinoModbus #163: API return code mismatch

bakadave, 2025-04-20:
I was debugging a failure to initialize registers when I stumbled upon this **major** documentation error.

From the api.md

> ### `modbusServer.configureHoldingRegisters()`
> [...]
> #### Returns
> 0 on success, 1 on failure

However in `ModbusServer.cpp` the function is defined as:
```
int ModbusServer::configureHoldingRegisters(int startAddress, int nb)
{
  if (startAddress < 0 || nb < 1) {
    errno = EINVAL;

    return -1;
  }

  size_t s = sizeof(_mbMapping.tab_registers[0]) * nb;

  _mbMapping.tab_registers = (uint16_t*)realloc(_mbMapping.tab_registers, s);

  if (_mbMapping.tab_registers == NULL) {
    _mbMapping.start_registers = 0;
    _mbMapping.nb_registers = 0;

    return 0;
  }

  memset(_mbMapping.tab_registers, 0x00, s);
  _mbMapping.start_registers = startAddress;
  _mbMapping.nb_registers = nb;

  return 1;
}
```

Which clerarly does the exact opposite.

per1234, 2025-04-20:
Hi @bakadave. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino-libraries/ArduinoModbus/issues/120.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

closed 2025-04-20 by per1234

bakadave, 2025-04-23:
got it, thx